demantz/RFAnalyzer #21: Maybe ADS-B and AIS support?

marksev1, 2015-03-11:
Would be really cool if one could display airplanes or ships on google maps :)...Is it in the pipeline or will it never happen? :D

demantz, 2015-03-11:
That's a cool idea! I love to track planes with the RTL SDR on my desktop. But I never thought of porting it to Android so far. But I think it should be a separate App. It doesn't really fit into RF Analyzer. I'm really short of time at the moment and this would be more like a toy app. But nevertheless it would be fun.

Android has a good API to use Google Maps inside Apps. But I think handling ads-b is not easy even if there are a couple of open source implementations are already out there.

I might start this as another project.. but at the moment there are too many other things around with higher priorities..
But thanks for the great idea!

marksev1, 2015-03-12:
No problem! All the current android ads-b apps are very limited at least for their free versions, so a new player in town would ne very welcome hehe :)..there also currently isn't an AIS app for android available. Cool lets se what the future brings :).

closed 2015-03-12 by marksev1