demantz/RFAnalyzer #22: bladerf??

jettero, 2015-03-23:
Seems to me this could be made to work with a bladerf.  I was slogging through https://github.com/demantz/hackrf_android/blob/master/src/com/mantz_it/hackrf_android/Hackrf.java with bpadalino on #bladerf and he thinks it's doable.

Is there support for others to add bindings? do you have any interest in coding it here?

I'm also opening an issue with nuand as bpadalino suggests maybe nuand would even write some bindings (as time allows, aka maybe not).

demantz, 2015-03-23:
I just had a quick look at libbladerf and I think it should be doable. Though to me it looks like the bladeRF lib is more complex than the hackRF lib. And of course we have to deal with 16bit (12bit) samples instead of the 8bit of the hackRF. And we can't get the full performance since Android does not support USB 3.0 (as far as I know).

However this still seems like a cool idea. Unfortunately I don't own a BladeRF yet and it's impossible to implement the Android bindings without hardware for testing ^^ I've already played with the idea of purchasing a BladeRF but so far it's out of my budget.. 

Maybe a BladeRF owner can help me with the Android bindings.. it shouldn't be too hard. Just by comparing these 3 files most of the questions should already be answered:
 https://github.com/demantz/hackrf_android/blob/master/src/com/mantz_it/hackrf_android/Hackrf.java
https://github.com/mossmann/hackrf/blob/master/host/libhackrf/src/hackrf.c
https://github.com/Nuand/bladeRF/blob/master/host/libraries/libbladeRF/src/backend/usb/libusb.c

Including the bindings into RF Analyzer should be easy..

wissanu020430, 2019-06-06:
https://github.com/demantz/RFAnalyzer/issues/69#issuecomment-499084698