demantz/RFAnalyzer #23: Can't load IQ file into SDR#

SonnyJim, 2015-04-11:
Just comes up with 'invalid file' :\

demantz, 2015-04-18:
That's correct. RF Analyzer produces files similar to hackrf_transfer -r (when used with a HackRF). These files are missing the WAV header which is required by SDR#. hackrf_transfer has the -w option to prepend this header. I might add this feature to RF Analyzer too. For now I just wrote a short python script that does the job: https://github.com/demantz/RFAnalyzer/blob/master/tools/prepWAV.py

I tested it with a files recorded by HackRF. I guess this doesn't work with RTL-SDR since the rtl-sdr driver outputs unsigned values (the HackRF outputs signed values). But I didn't tested it. Does anybody know which values to write in the WAV header in order to state that it contains unsigned bytes?

Anyway I hope this helps!

jamesweech, 2015-04-18:
Works fine on RTL SDR, Please build into program though as its a bit of a PITA to convert!

Thanks so much :)

jamesweech, 2015-04-18:
Note on the RTL SDR on SDR# it does not show the frequency only 0.0

wissanu020430, 2019-06-06:
https://github.com/demantz/RFAnalyzer/issues/69#issuecomment-499084698

lucapgt, 2019-07-29:
Hi, I have this error:

Write WAV header...
Traceback (most recent call last):
  File "prepWAV.py", line 65, in <module>
    writeWavHeader(outFile, filesize, args.samplerate)
  File "prepWAV.py", line 21, in writeWavHeader
    file.write(struct.pack('4s', "RIFF"))
struct.error: argument for 's' must be a bytes object

Snowolf842, 2019-08-07:
Hello,
I had the same error as you, use python2 to launch the script, it worked for me :)