demantz/RFAnalyzer #24: works with rtl-sdr, fails with HackRF (on HTC One)

ecc1, 2015-06-23:
I'm trying to use my HackRF on my first-generation HTC One (M7?). When I set the source to HackRF and press start, I first get a popup that says it found the HackRF at a certain USB address, but a few seconds later it stops with a popup that says "Error: HackRF stopped receiving".  That happens each time I press start.  When I use an rtl-sdr dongle, it works fine.

demantz, 2015-06-24:
Thanks for the issue report!

Hmm I digged through my mail and couldn't find anything about the M7. Maybe no one tried it before? But I know it works on an M8 (see list of supported devices: https://github.com/demantz/hackrf_android#testet-devices (incomplete)).

Can you send me the log file of RF Analyzer after the error occured? You have to enable logging first (goto settings, advanced settings and activate logging). Maybe I can find some additional information about the error in the log.

If you are willing to do some more experiments I would ask you to install the test app of the hackrf_android driver (https://github.com/demantz/hackrf_android/blob/master/examples/HackRF_Test.apk?raw=true). With this app you will see a bit more information of what happens when the HackRF stops receiving. You can also try different sample rates. Here is a link to an old post on my blog where you can see how the app is used: http://tech.mantz-it.com/2014/10/hackrfandroid-using-hackrf-with-android.html

Cheers,
Dennis

ecc1, 2015-06-24:
On Wed, Jun 24, 2015 at 06:48:29AM -0700, Dennis Mantz wrote:
> Can you send me the log file of RF Analyzer after the error occured? You have to
> enable logging first (goto settings, advanced settings and activate logging).
> Maybe I can find some additional information about the error in the log.

I've attached a log.  This is from two consecutive runs.  Both of
these lasted longer than before, and didn't fail until I also turned
on demodulation (Previously it would fail even with demodulation off,
after a few seconds.  I don't know what changed.)  In the first run, I
left the initial frequency (91.3MHz) unchanged.  In the second, I
tuned around to a different station before turning on modulation.

> If you are willing to do some more experiments I would ask you to install the
> test app of the hackrf_android driver
> (https://github.com/demantz/hackrf_android/blob/master/examples/HackRF_Test.apk?
> raw=true). With this app you will see a bit more information of what happens
> when the HackRF stops receiving. You can also try different sample rates. Here
> is a link to an old post on my blog where you can see how the app is used:
> http://tech.mantz-it.com/2014/10/hackrfandroid-using-hackrf-with-android.html

Thanks, I'll try this.

--
Eric Cooper             e c c @ c m u . e d u

--------- beginning of main
I/art     (24282): Late-enabling -Xcheck:jni
I/art     (24282): VMHOOK: rlim_cur : 4294967295 pid:24282
W/Resources(24282): Converting to string: TypedValue{t=0x10/d=0x0 a=-1}
W/Resources(24282): Converting to string: TypedValue{t=0x10/d=0x0 a=-1}
D/OpenGLRenderer(24282): Use EGL_SWAP_BEHAVIOR_PRESERVED: true
D/qdmemalloc(24282): ion: Mapped buf base:0xa401c000 size:14221312 offset:0 fd:27
D/qdmemalloc(24282): ion: Mapped buf base:0xa401b000 size:4096 offset:0 fd:28
D/Atlas   (24282): Validating map...
W/Resources(24282): Converting to string: TypedValue{t=0x10/d=0x0 a=-1}
W/Resources(24282): Converting to string: TypedValue{t=0x10/d=0x0 a=-1}
I/Adreno-EGL(24282): <qeglDrvAPI_eglInitialize:410>: EGL 1.4 QUALCOMM build: AU_LINUX_ANDROID_LA.AF.1.1_RB1.05.00.02.006.018_msm8960_LA.AF.1.1_RB1__release_AU ()
I/Adreno-EGL(24282): OpenGL ES Shader Compiler Version: E031.25.03.00
I/Adreno-EGL(24282): Build Date: 02/12/15 Thu
I/Adreno-EGL(24282): Local Branch: mybranch7569669
I/Adreno-EGL(24282): Remote Branch: quic/LA.AF.1.1_rb1.12
I/Adreno-EGL(24282): Local Patches: NONE
I/Adreno-EGL(24282): Reconstruct Branch: AU_LINUX_ANDROID_LA.AF.1.1_RB1.05.00.02.006.018 +  NOTHING
I/OpenGLRenderer(24282): Initialized EGL, version 1.4
D/OpenGLRenderer(24282): Enabling debug mode 0
D/qdmemalloc(24282): ion: Mapped buf base:0xa328a000 size:8355840 offset:0 fd:36
D/qdmemalloc(24282): ion: Mapped buf base:0xb3d98000 size:4096 offset:0 fd:37
D/qdmemalloc(24282): ion: Mapped buf base:0xa2a92000 size:8355840 offset:0 fd:39
D/qdmemalloc(24282): ion: Mapped buf base:0xb3cf1000 size:4096 offset:0 fd:40
D/qdmemalloc(24282): ion: Mapped buf base:0xa20d3000 size:8355840 offset:0 fd:41
D/qdmemalloc(24282): ion: Mapped buf base:0xb0d46000 size:4096 offset:0 fd:42
D/qdmemalloc(24282): ion: Mapped buf base:0xa1735000 size:8355840 offset:0 fd:44
D/qdmemalloc(24282): ion: Mapped buf base:0xb0d3a000 size:4096 offset:0 fd:45
D/qdmemalloc(24282): ion: Mapped buf base:0xa0f3d000 size:8355840 offset:0 fd:46
D/qdmemalloc(24282): ion: Mapped buf base:0xa0f3c000 size:4096 offset:0 fd:47
D/qdmemalloc(24282): ion: Mapped buf base:0xa0744000 size:8355840 offset:0 fd:48
D/qdmemalloc(24282): ion: Mapped buf base:0xa0743000 size:4096 offset:0 fd:49
W/KeyCharacterMap(24282): Load KCM of non-default device may incur unexpected result. Device ID:9
D/OpenGLRenderer(24282): Flushing caches (mode 0)
D/qdmemalloc(24282): ion: Unmap buf  base:0xa1735000 size:8355840
D/qdmemalloc(24282): ion: Unmap buf  base:0xb0d3a000 size:4096
D/qdmemalloc(24282): ion: Unmap buf  base:0xa0f3d000 size:8355840
D/qdmemalloc(24282): ion: Unmap buf  base:0xa0f3c000 size:4096
D/qdmemalloc(24282): ion: Unmap buf  base:0xa0744000 size:8355840
D/qdmemalloc(24282): ion: Unmap buf  base:0xa0743000 size:4096
D/OpenGLRenderer(24282): Flushing caches (mode 0)
W/InputEventReceiver(24282): Attempted to finish an input event but the input event receiver has already been disposed.
W/KeyCharacterMap(24282): Load KCM of non-default device may incur unexpected result. Device ID:9
I/MainActivity(24282): onCreate: started logcat (Process[pid=24493]) to /storage/emulated/0/RFAnalyzer/log.txt
I/AnalyzerSurface(24282): setFontSize: X-dpi=472.965 X-width=1080  fontSize=2  normalTextSize=84  smallTextSize=44
[... 555 lines cut ...]
D/qdmemalloc(24282): ion: Unmap buf  base:0xa2c41000 size:737280
D/qdmemalloc(24282): ion: Unmap buf  base:0xb3cfd000 size:4096
D/OpenGLRenderer(24282): Flushing caches (mode 0)
E/hackrf_android(24282): Couldn't claim HackRF USB Interface!
E/HackRFSource(24282): stopSampling: Error while tear down hackrf: Couldn't claim HackRF USB Interface!
D/qdmemalloc(24282): ion: Unmap buf  base:0xa0290000 size:3690496
D/qdmemalloc(24282): ion: Unmap buf  base:0xb3cdc000 size:4096
D/qdmemalloc(24282): ion: Unmap buf  base:0x9f87e000 size:3690496
D/qdmemalloc(24282): ion: Unmap buf  base:0xaf781000 size:4096
D/qdmemalloc(24282): ion: Unmap buf  base:0x9fc03000 size:3690496
D/qdmemalloc(24282): ion: Unmap buf  base:0xb0d3a000 size:4096
D/OpenGLRenderer(24282): Flushing caches (mode 0)
D/qdmemalloc(24282): ion: Unmap buf  base:0xa3079000 size:8355840
[... cut ...]
D/qdmemalloc(24282): ion: Unmap buf  base:0xa0a43000 size:8355840
D/qdmemalloc(24282): ion: Unmap buf  base:0xb3d03000 size:4096
D/OpenGLRenderer(24282): Flushing caches (mode 0)
D/OpenGLRenderer(24282): Flushing caches (mode 1)
D/OpenGLRenderer(24282): Flushing caches (mode 0)
D/OpenGLRenderer(24282): Flushing caches (mode 0)
V/UsbRequest(24282): endpoint still open in finalize(): android.hardware.usb.UsbRequest@2c29f817
D/UsbRequestJNI(24282): close
D/OpenGLRenderer(24282): Flushing caches (mode 0)

ecc1, 2015-06-24:
I'm embarrassed to say that I think it was my USB cable.  I was
switching several of them around:
  - on #1, the HackRF source is not detected at all
  - on #2, it starts OK but fails soon with the "stopped receiving"
  error
  - on #3, it's been working fine with no failures

If the app can detect anything marginal in case #2, perhaps a warning
message would be helpful, but otherwise I think you can close this
issue and consider it working on the HTC One M7.  Thanks again.

--
Eric Cooper             e c c @ c m u . e d u

ckuethe, 2015-06-24:
The hackrf works with all those cables when plugged into a PC, right? Maybe some of the cables have enough of a difference in resistance that they are unable to power the hackrf properly.

demantz, 2015-06-26:
Yes, faulty / bad quality USB cables or OTG adapters seem to be pretty common. I've heard from a lot of people having power problems... I'll add a warning to the Readme and the Play Store!

Anyway I'm glad it's working for you now!

closed 2015-06-26 by demantz