demantz/RFAnalyzer #25: Enhancement request: Support for RSP (SDRPlay)

dc1rdb, 2015-07-18:
The RSP (Radio Spectrum Processor) from SDRplay is a Software Defined Radio covering 100kHz to 2GHz with a 12bit ADC.
API and Android integration instructions can be found at http://sdrplay.com/api_drivers.html

demantz, 2015-07-20:
Thanks for this suggestion! This is actually the first time I heard of the RSP, but it sounds like an interesting device. According to the API specifications it will output IQ samples which are required by RF Analyzer.

I don't know where I could borrow a RSP and currently I can't afford buying it. I will keep this issue open and maybe add support for the RSP when I get access to a device somehow!

dc1rdb, 2015-07-21:
Thanks Dennis! I'd be happy to do some testing for you - just let me know! Best regards, Robert

dc1rdb, 2015-09-22:
Alternatively, you might want to consider using the SoapySDR library to easily incorporate all different kinds of RF frontends / dongles. Details can be found at https://github.com/pothosware/SoapySDR/wiki . Driver modules for RTLSDR, SDRPlay and a couple of other SDRs are already available.

He3556, 2015-10-20:
I just got a RTL-SDR for researching. 
We will use some kind of SDR /RSP in the future. If somebody has experience how to get the values out of the decoded stream and put it in a db - please let us know.

wissanu020430, 2019-06-06:
https://github.com/demantz/RFAnalyzer/issues/69#issuecomment-499084698