demantz/RFAnalyzer #29: Problem with RFAnalyzer

reclaudio, 2016-01-07:
Dear Dennis ,
                      I have bought your app RFAnalyzer from Play Store , connected trough an OTG cable to HackRF that is correctly powered from the smartphone when plugged in .If I start your app it appears : Source not available (HackRF) . The Smartphone il Alcatel One Touch IDOL X 6043D . Android version 4.4.2 Kernel 3.4.67 .Could you please help me ? I have also read while buying on Play Store that could also work with SDR dongles . I tried one but don't even power on .
Thank you and kind regards .
Claudio  Re 
reclaudio@alma.it

demantz, 2016-01-08:
Hello Claudio,

your setup looks fine to me. There are only two possibilities I can think of which could explain why you get an error:

1) Your Smartphone does not support USB OTG. Can you verify this works (e.g. by plugging in an USB thumb drive and see if you can read/write it)? OTG may not be supported (either by the hardware or by the Kernel).
2) The HackRF might not get enough power (some phones can not supply enough power on their USB port or the USB cable is bad). In this case it is possible that the HackRF LEDs light up but as soon as the Phone tries to communicate with the HackRF the power supply breaks down. But this usually results in another kind of error message.

Other than that I'm afraid I can't help you. If the app can't find the HackRF it means that no USB device with a matching ID is listed. You can look into the log file of RF Analyzer to see which USB devices are listed (I guess there are no devices listed). You can also try to use other apps (like https://play.google.com/store/apps/details?id=com.synchack.android.usbhostviewer) to see which USB devices are listed.

Good luck with solving this issue! Please keep me up to date if you find out anything new!

Cheers,
Dennis

reclaudio, 2016-01-08:
Hello Dennis ,

                           I see USB key dongles .

Wil try to see with  <https://play.google.com/store/apps/details?id=com.synchack.android.usbhostviewer> https://play.google.com/store/apps/details?id=com.synchack.android.usbhostviewer , but if I don’t see anything , like I am afraid , what can I do …

Thank you and kind regards .

Claudio

A: demantz/RFAnalyzer
Cc: reclaudio
Oggetto: Re: [RFAnalyzer] Problem with RFAnalyzer (#29)

Questa e-mail è stata controllata per individuare virus con Avast antivirus.
https://www.avast.com/antivirus

This message has been scanned by E.F.A. Project and is believed to be clean.

reclaudio, 2016-01-09:
Hello Dennis .
The log showed to try to control the HackRF , but with no response from the device .

Will try again with perhaps a more smart OTG that I have ordered .

This app was not compatible with my Smart Phone

 <https://play.google.com/store/apps/details?id=com.synchack.android.usbhostviewer> https://play.google.com/store/apps/details?id=com.synchack.android.usbhostviewer.

At the end I was able to make it work with a Nexus 7 like yours .

Please see here the publication of the results :

http://air-radiorama.blogspot.it/2016/01/tablet-android-hackrf-rfanalyzer-rx.html

BTW : will you be at RadioHack ?

I will .

If yes, see you there .

Thank you and kind regards .

Claudio Re

Da: Claudio Re [mailto:reclaudio@alma.it]
Inviato: venerdì 8 gennaio 2016 10:17
A: 'demantz/RFAnalyzer'; 'demantz/RFAnalyzer'
Oggetto: R: [RFAnalyzer] Problem with RFAnalyzer (#29) A
Priorità: Alta

Hello Dennis ,

                           I see USB key dongles .

Wil try to see with  <https://play.google.com/store/apps/details?id=com.synchack.android.usbhostviewer> https://play.google.com/store/apps/details?id=com.synchack.android.usbhostviewer , but if I don’t see anything , like I am afraid , what can I do …

Thank you and kind regards .

Claudio

https://www.avast.com/antivirus

wissanu020430, 2019-06-06:
https://github.com/demantz/RFAnalyzer/issues/69#issuecomment-499084698

closed 2025-08-05 by demantz