demantz/RFAnalyzer #31: SecurityException on Samsung S5

demantz, 2016-01-21:
So far only one user reported this bug on the Samsung S5, but according to several boards in the internet it affects all Samsung S5 (since the Lollipop update) users and all apps which use external USB devices:

http://developer.samsung.com/forum/board/thread/view.do?boardName=General&messageId=285782

The exception occurs when trying to request the permissions to connect to the USB device:

    java.lang.SecurityException: Admin  does not have android.permission.sec.MDM_APP_MGMT
        at android.os.Parcel.readException(Parcel.java:1546)
        at android.os.Parcel.readException(Parcel.java:1499)
        at android.hardware.usb.IUsbManager$Stub$Proxy.requestDevicePermission(IUsbManager.java:653)
        at android.hardware.usb.UsbManager.requestPermission(UsbManager.java:432)
        at com.mantz_it.hackrf_android.Hackrf.initHackrf(Hackrf.java:228)
        at com.mantz_it.rfanalyzer.HackrfSource.open(HackrfSource.java:88)

It also seems like there is no workaround:
https://github.com/saki4510t/UVCCamera/issues/20

Lets wait for Samsung to fix this annoying bug.

jesmith0, 2017-02-21:
S5 user here. Experiencing the same issue. Just wanted to add my name to the list.

abhishekkumar1902, 2017-08-26:
Something strange is happening with my case. I am using Samsung note edge running Android v5.1.1. I, where I compiled all the 9 uvccamera projects and tested on the phone and it was successfully able to detect the camera and fetch the video from the camera as well. Now, just what I did is just copied the same code and layout widget design from **usbCameraTest0** project line by line without any modification and integrated it into a new project targeted at **same compiler, API and SDK level**. There happens no problems in compilation and starting of this new app but the problem is that now when I press the "connect to" camera button, the dialog box comes in but it does not detect the camera at all. Please help me with this issue. At the same time, if I run the usbCameraTest0 project alone, it runs fine. Not sure what could be wrong. Please help.

wissanu020430, 2019-06-06:
https://github.com/demantz/RFAnalyzer/issues/69#issuecomment-499084698