demantz/RFAnalyzer #33: RTL SDR in Direct Sampling Mode?

purirohan, 2016-04-11:
Any recommendations on how to modify the source to support direct sampling as mentioned in articles like this:

http://www.rtl-sdr.com/rtl-sdr-direct-sampling-mode/

and this (translate from Japanese and it's still quite readable, see the line "You specify the parameters for use in the direct sampling open the configuration screen of the I / O Devices from the menu. In gqrx and gnuradio in Device string, and then append the direct_samp = 1. How to specify this are common in the software you are using librtl."):

http://ttrftech.tumblr.com/post/36283680013/rtl2832u-dongle-mods-for-7mhz-reception-with-gqrx

Wondering if there's an equivalent in the code you developed! Thanks so much :+1:

demantz, 2016-04-17:
Hmm, this setting would probably have to be set in the RTL2832U driver I'm using in RF Analyzer. However, I'm not sure if it is possible and how to do it, sorry. Maybe I'll look into this, but currently I don't have enough time to spend..

JonTabor, 2017-03-07:
I would love this feature; SDR Touch has it, but of course it's $9.99 to enable all the features there.

I'll go digging into the code and see what I can find.

jbvoelker, 2017-05-11:
I have implemented basic support for the direct sampling mode [in my fork](https://github.com/jbvoelker/RFAnalyzer/commit/cfad9e6a096a322129108d5dbd6a27af9186728f) of RFAnalyzer.
Off, I branch and Q branch can be selected via the preferences and become active after restarting the sampling.

I have tried fitting it into the existing architecture but the change is currently a "works for me" version.
I also didn't rebuild the apk.

JonTabor, 2017-05-11:
I pulled your fork, built and tested.  Appears to work well on two different RTL SDR dongles (R820T and R820T2-based, generic eBay units) on my Kindle Fire, running Android 5.1.1 (rooted/custom ROM - none of that FireOS stuff).  

Nicely done!  And thank you!

imrekisslyb, 2017-08-18:
Could you please share the built APK as well?

jbvoelker, 2017-09-13:
@imrekisslyb I have added a build with my changes in my fork (RFAnalyzer.apk).

closed 2026-03-15 by demantz