demantz/RFAnalyzer #36: High frequency noise

truthz03, 2016-04-30:
I have tried my RTL-SDR stick with our software and it worked great but I found out that there is a background noise with a very high frequency.
I also tried the stick with SDRTouch and with that app this noise is not present.
I don't know if that noise could be the rds data or other overlaying informations or maybe only a filter is needed.

Thanks Thomas

wissanu020430, 2019-06-06:
https://github.com/demantz/RFAnalyzer/issues/69#issuecomment-499084698

demantz, 2025-08-04:
SDRTouch and RF Analyzer use the same driver to get the signal from the RTL-SDR. I could imaging that SDRTouch does a bit of averaging to smooth the display of the FFT. RF Analyzer also has a setting for averaging, maybe give it a try?