demantz/RFAnalyzer #38: Are floating point calculations really necessary?

pavlus, 2016-09-04:
As soon as AudioSink converts `float` samples to `short`, can't all processing be rewritten using integer numbers to increase performance, or there is something that forces to use floats?

demantz, 2016-09-11:
Yeah I think you are correct on this. If the developer guide is correct, floating point operation are about two times slower than integer (https://developer.android.com/training/articles/perf-tips.html#AvoidFloat).

Initially I implemented RF Analyzer without demodulation support and I chose floating point calculations out of convenience. Now, with demodulation running, it would be great to have this additional bit of performance, but I never found the time to do the refactoring. 

I see you are doing quite some interesting modifications on the RF Analyzer code.. if you have the time please go ahead and switch to integer arithmetic! If the performance boost is significant I would love to pull these changes back into my repo!

Cheers,
Dennis

pavlus, 2016-09-12:
I'm extending your project for my master's thesis to support (clone of) HiQSDR direct conversion receiver, but differences in characteristics exposed some design nuances which lead to these modifications. For example integer-float conversion, I thought it would be a bottleneck for this device, but after measuring direct type conversion with bit operations and floating point multiplication was as fast as 8-bit LUT, so I'm not planning to rewrite everything to integers in first order. 

Now I'm digging in decimator-demodulator pipeline and need to make it more flexible to support different samplerates (for example, HiQSDR lowest rate is 48ksps, we can bypass decimation).

Creating flexible and fast decimation pipeline is my first order task for now. Wide Android AudioTrack samplerate support (4-96kHz) lets us convert to different final samplerates, so we just need to find optimal decimation value and dynamically design FIR filter for it

I've looked at your DSP lib and think that it will make much more boost to performance than integer math(based on benchmark results), but couldn't import it to AndroidStudio to play with it.

Respectfully,
Pavel.

wissanu020430, 2019-06-06:
https://github.com/demantz/RFAnalyzer/issues/69#issuecomment-499084698

closed 2025-09-26 by demantz