demantz/RFAnalyzer #39: "Error: HackRF stopped receiving".

i02much, 2016-09-28:
I've been fighting this error for weeks.  I'm way out of my league trying to compile apk files and can barely spell Got Hub...  I saw a closed post on this issue where the guy was having problems with bad OTG / USB cables.  I've got 3 new OTG (a variety) and several USB cables and I have the same problem with each one
  I've tried different settings.  It will play great for a minute and the audio will start skipping and the receive light will go out on the HackRF and Ill get the   "Error: HackRF stopped receiving".   I can try to upload the log

i02much, 2016-09-28:
--------- beginning of main
09-16 15:35:42.978  4812  6626 I chatty  : uid=10101(com.mantz_it.rfanalyzer) expire 23 lines
09-16 15:35:53.200  4812  6626 I chatty  : uid=10101(com.mantz_it.rfanalyzer) expire 41 lines
09-16 15:36:04.301  4812  6626 I chatty  : uid=10101(com.mantz_it.rfanalyzer) expire 57 lines
09-16 15:36:21.021  4812  6626 I chatty  : uid=10101(com.mantz_it.rfanalyzer) expire 7 lines
09-16 15:36:22.569  4812  6629 I chatty  : uid=10101(com.mantz_it.rfanalyzer) expire 1 line
09-16 15:36:26.601  4812  7454 I chatty  : uid=10101(com.mantz_it.rfanalyzer) expire 4 lines
09-16 15:36:26.603  4812  7455 I chatty  : uid=10101(com.mantz_it.rfanalyzer) expire 17 lines
09-16 15:36:26.605  4812  7456 I chatty  : uid=10101(com.mantz_it.rfanalyzer) expire 1 line
09-16 15:36:26.608  4812  7458 I chatty  : uid=10101(com.mantz_it.rfanalyzer) expire 3 lines
09-16 15:36:26.612  4812  7460 I chatty  : uid=10101(com.mantz_it.rfanalyzer) expire 2 lines
09-16 15:36:26.612  4812  7459 I chatty  : uid=10101(com.mantz_it.rfanalyzer) expire 3 lines
09-16 15:36:40.915  4812  7455 I chatty  : uid=10101(com.mantz_it.rfanalyzer) expire 3 lines
09-16 15:36:41.551  4812  4812 I chatty  : uid=10101(com.mantz_it.rfanalyzer) expire 6 lines
09-16 15:36:43.049  4812  7459 I chatty  : uid=10101(com.mantz_it.rfanalyzer) expire 1 line
09-16 15:36:43.504  4812  7455 I chatty  : uid=10101(com.mantz_it.rfanalyzer) expire 1 line
09-16 15:36:43.891  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:44.509  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:45.023  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:45.855  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:46.399  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:46.772  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:47.108  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:47.634  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:47.989  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:49.227  4812  7459 W AudioTrack: releaseBuffer() track 0x7f94e96780 disabled due to previous underrun, restarting
09-16 15:36:50.114  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:50.532  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:51.024  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:51.515  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:51.921  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:52.234  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:52.490  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:52.777  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:52.988  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:53.457  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:53.672  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:53.773  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:55.070  4812  7459 W AudioTrack: releaseBuffer() track 0x7f94e96780 disabled due to previous underrun, restarting
09-16 15:36:55.505  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:55.841  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:56.176  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:56.521  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:56.856  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:57.192  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:57.536  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:57.872  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:58.208  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:58.552  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:58.888  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:59.224  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:59.568  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:36:59.904  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:37:00.232  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
[... cut ...]
09-16 15:37:18.000  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:37:18.336  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:37:18.800  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:37:19.122  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:37:19.343  4812  7455 D Scheduler: run: Flush the demod queue because demodulator is too slow!
09-16 15:37:19.871  4

demantz, 2016-09-28:
hmm wow .. it seems like the receiving queue runs full almost immediately.. may I ask which android device you are using?

i02much, 2016-10-03:
I'm using a Nexus 6P running Resurrection Remix 6.01.01 ROM.

I can send more details if needed, just let me know.  

Thank you! I absutely love this app.  BTW it works flawlessly when I connect my RTLSDR
I'm only getting the error when using it with HackRF.

Sent from BlueMail

i02much, 2016-10-08:
OK. So I picked up another OTG cable that fits the wife's Nexus 6 and installed RFAnalyzer on it just to see and I get the same error.  I just bought the HackRF less than a month ago.  Could it be a hardware problem? It works fine with SDR# on the PC.  Just has problems on Android...

alkimpone, 2017-02-16:
I do have same error on my Honor 7 when i activate modulation , all work fine when modulation mode is off  , but instant after i turn to fm or am or any other the app start lagg as crazy same with sound and image and i getting same error message :/ 
Also got this message when i been changing frequency to fast or zoomed to long out ,
Also if i continue the app will crash and turn white everytime i reopen it , even when its closed in app manager , and on rebooting my phone hangs up so have to hardreset to reset my phone , do belive it have somthing to do with the OTG unexpected getting disconnected or somthing from the device and make the system hang maybe ?
Using rooted phone with android version : 5.0.2

pavlus, 2017-02-17:
Can you record signal to file, play it back on repeat and check if problem persist?

DavidVanHorn, 2019-01-31:
I have seen this problem as well. It seems to be tied to certain frequency ranges, but I am having trouble finding how to replicate the problem repeatably.  LG V30+ here, with HackRF one, and 6" cable.

wissanu020430, 2019-06-06:
https://github.com/demantz/RFAnalyzer/issues/69#issuecomment-499084698