demantz/RFAnalyzer #41: To enhance performance by change Java FFTs to native FFTs

james34602, 2016-10-19:
My experience tell that use a high performance native FFT libraries will grant performance.
The following repository is my experiment on two different microphone/custom signal source audio spectrum analyzer application in Android, both use native method to perform FFT, one of them use native to perform extra audio filtering, especially repository one have significant performance gain from change KissFFT to Ne10 FFT.
My concern is slow FFT will use more CPU cycle to perform calculations, hence drain battery faster.
1. https://github.com/james34602/AudioAnalyzer
2. https://github.com/james34602/audio-analyzer-for-android
Compile tested and confirm worked on armeabi-v7a(with NEON/without) and x86 platform.

aneitzel, 2016-10-19:
Hi,

sounds good! Is a android APK avialable for download?

james34602, 2016-10-20:
Yeah, I uploaded it.
1. https://github.com/james34602/AudioAnalyzer/releases
2. https://github.com/james34602/audio-analyzer-for-android/releases

pavlus, 2016-10-21:
Actually, there is a project by Dennis Mantz to implement DSP on RenderScript (https://github.com/demantz/android_dsp_lib), which may increase perfromance even more than native libraries, but it currently doesn't support FFT. I'm going to work on FFT and some more filters support with it in my fork, when I'm done with HiQSDR support and reworked pipeline for decimation-demodulation.

james34602, 2016-10-22:
I hope will see a FFT using RenderScript, graphic display using RS is great, but use it as 'standard'  computation, which performance gain maybe controversy and device depend.
But I will try to modify RFAnalyzer into native FFT calculation.

pavlus, 2016-10-22:
yes, that's why i'm trying to rewrite current code to be more independent from other parts, I hope, I'll be able to to maximally "pluginize" app, so user just could install library on their phone and new options appear (I'd like things like opengl renderer, different realizations of dsp, user filters, demodulators, decoders, etc), so user could just install your lib, go to settings and select "Native FFT"

demantz, 2025-08-05:
RF Analyzer 2.0 does FFT in native code and allows much higher FFT Size values.

closed 2025-08-05 by demantz