demantz/RFAnalyzer #43: RTL2832U driver not installed

atpage, 2016-12-14:
I am getting this error when attempting to use RF Analyzer on a Nexus 6 with Android 6.0.1.  The source is set to RTL-SDR.  The driver is installed and it works in SDRTouch.

mvglasow, 2017-04-03:
Same here. When I choose "Install from Google Play", I get the F-Droid entry for the driver (which correctly tells me it’s installed).

mvglasow, 2017-04-04:
I’ve found something. Turns out F-Droid has an outdated driver version (they can’t build the current one). After replacing it with the current APK from the dev’s git repo, the message disappears. Don’t have my stick at hand so I can’t fully test it, will do so as soon as I get to it.

demantz, 2017-04-04:
Sorry for the late reply, seems like mvglasow has already solved it:
The RTL2832U version on F-Droid is from 2014. The API of this driver has changed since then and the latest RF Analyzer is only compatible with the newer versions. As a solution: Try to find the latest version of the RTL2832U apk somewhere online and don't use the old version from F-Droid. Thanks to mvglasow!

closed 2017-04-04 by demantz

mvglasow, 2017-04-04:
In the meantime I’ve opened https://github.com/martinmarinov/rtl_tcp_andro-/issues/12 in the hope that the F-Droid issues are going to get fixed.

Btw, the project offers prebuilt APKs in their repo at https://github.com/martinmarinov/rtl_tcp_andro-/blob/master/app/app-release.apk. These are current and should work.