demantz/RFAnalyzer #45: doesnt work on the MTK6592 Android4.4.2

dextercai, 2016-12-18:
yep, the bata version doesnt work on the android4.4.2 and the cpu is mtk6592.
and use the version 1.13 will always make phone reboot.

pavlus, 2016-12-21:
Can you attach logs?

dextercai, 2016-12-22:
Actually not, but I have kernel log

dextercai, 2016-12-23:
Kernel (KE),0,0,99,/data/core/,0,,KE at musb_h_disable,Sun Dec 11 08:23:39 CST 2016,1
here is the kernel log.
[KE.dbg.zip](https://github.com/demantz/RFAnalyzer/files/670799/KE.dbg.zip)

pavlus, 2016-12-24:
It's password-protected. No use of it. How about, if you open adb shell on pc, run `cat /proc/kmsg` (as SU), provoke reboot by app and paste part of log that appeared after starting app?

Have you tried previous versions? It crashes on start or when?

dextercai, 2016-12-24:
well, just like the info given before. This is happening when the musb_h_disable.when app run about 30 seconds on my phone. The application will be stuck. if l try to close the connection between app and hackrf and reconnect them, it doesnt works. if l pull USB  out, it will stuck some seconds and reboot. 

the apk version is the 1.13alpha3, the other version cant use one my phone especially 1.13(1303), it always FC when l open it.

l will try to catch the log and paste it later.

dextercai, 2016-12-25:
root@lcsh92_wet_tdd:/storage/emulated/0 # cat /proc/last_kmsg
ram console header, hw_status: 2, fiq step 40.
bin log 0.
i]  .rt_time                       : 0.000000
[42476.987185] (0)[848:ndroid.systemui]  .rt_runtime                    : 950.000000
[42476.987198] (0)[848:ndroid.systemui]
[42476.987203] (0)[848:ndroid.systemui]runnable tasks:
[42476.987209] (0)[848:ndroid.systemui]            task   PID         tree-key  switches  prio     exec-runtime         sum-exec        sum-sleep
[42476.987218] (0)[848:ndroid.systemui]----------------------------------------------------------------------------------------------------------
[42476.997469] (0)[848:ndroid.systemui]==========================================================================================================
[42476.997484] (0)[848:ndroid.systemui]
[42476.997649] (0)[848:ndroid.systemui]-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
[42486.915570] (0)[848:ndroid.systemui][aee/ipanic] aee-ipanic: Panic dump sucessfully written to emmc (detail len: 19648, console len: 262144)
[42486.915595] (0)[848:ndroid.systemui][aee/ipanic] android log : 0x4e000, 0x40000, 0x0, 0x0, 0x8e000, 0x40000, 0xce000, 0xa000
[42486.915612] (0)[848:ndroid.systemui][aee/ipanic] mmprofile: offset:0xd8000, len:0x0
[42486.915628] (0)[848:ndroid.systemui]Rebooting in 1 seconds..machine_shutdown: start, Proess(ndroid.systemui:848)
[42487.909471] (0)[848:ndroid.systemui]Backtrace:
[42487.909500] (0)[848:ndroid.systemui][<c00128c4>] (dump_backtrace+0x0/0x114) from [<c07d2cf8>] (dump_stack+0x20/0x24)
[42487.909516] (0)[848:ndroid.systemui] r6:000003e8 r5:00000000 r4:000003e8 r3:c0b22738
[42487.909559] (0)[848:ndroid.systemui][<c07d2cd8>] (dump_stack+0x0/0x24) from [<c000fc90>] (machine_shutdown+0x3c/0x58)
[42487.909587] (0)[848:ndroid.systemui][<c000fc54>] (machine_shutdown+0x0/0x58) from [<c000fd20>] (machine_restart+0x1c/0x6c)
[42487.909616] (0)[848:ndroid.systemui][<c000fd04>] (machine_restart+0x0/0x6c) from [<c006f858>] (emergency_restart+0x24/0x28)
[42487.909631] (0)[848:ndroid.systemui] r5:00000000 r4:000003e8
[42487.909661] (0)[848:ndroid.systemui][<c006f834>] (emergency_restart+0x0/0x28) from [<c07d3208>] (panic+0x140/0x1a8)
[42487.909686] (0)[848:ndroid.systemui][<c07d30c8>] (panic+0x0/0x1a8) from [<c0012bf0>] (die+0x128/0x2b4)
[42487.909700] (0)[848:ndroid.systemui] r3:00010000 r2:d106bb78 r1:20000193 r0:c095cac4
[42487.909730] (0)[848:ndroid.systemui] r7:00000001
[42487.909755] (0)[848:ndroid.systemui][<c0012ac8>] (die+0x0/0x2b4) from [<c07d2d60>] (__do_kernel_fault.part.10+0x64/0x84)
[42487.909784] (0)[848:ndroid.systemui][<c07d2cfc>] (__do_kernel_fault.part.10+0x0/0x84) from [<c0016f80>] (do_page_fault+0x22c/0x278)
[42487.909800] (0)[848:ndroid.systemui] r7:dbec0a80 r3:d106bd88
[42487.909832] (0)[848:ndroid.systemui][<c0016d54>] (do_page_fault+0x0/0x278) from [<c0017104>] (do_translation_fault+0xa8/0xb0)
[42487.909859] (0)[848:ndroid.systemui][<c001705c>] (do_translation_fault+0x0/0xb0) from [<c00083d4>] (do_DataAbort+0x80/0x118)
[42487.909874] (0)[848:ndroid.systemui] r7:c0b17014 r6:d106a000 r5:00000005 r4:d106bd88
[42487.909915] (0)[848:ndroid.systemui][<c0008354>] (do_DataAbort+0x0/0x118) from [<c000e4d8>] (__dabt_svc+0x38/0x60)
[42487.909931] (0)[848:ndroid.systemui]Exception stack(0xd106bd88 to 0xd106bdd0)
[42487.909949] (0)[848:ndroid.systemui]bd80:                   00000000 c9891d80 dee5c44c c7986790 c9891d80 dee5c0d8
[42487.909969] (0)[848:ndroid.systemui]bda0: 00000080 dee5c0ec 80000193 c7986780 c7986790 d106bdfc d106bda0 d106bdd0
[42487.909986] (0)[848:ndroid.systemui]bdc0: c07df1b8 c0415f2c 60000193 ffffffff
[42487.909998] (0)[848:ndroid.systemui] r8:80000193 r7:d106bdbc r6:ffffffff r5:60000193 r4:c0415f2c
[42487.910046] (0)[848:ndroid.systemui][<c0415e28>] (musb_h_disable+0x0/0x160) from [<c0419ac8>] (musb_h_pre_disable+0x6c/0xe0)
[42487.910071] (0)[848:ndroid.systemui][<c0419a5c>] (musb_h_pre_disable+0x0/0xe0) from [<c0410bec>] (musb_root_disconnect+0x5c/0x18c)
[42487.910087] (0)[848:ndroid.systemui] r8:00000000 r7:df1fbf80 r6:c0e8db5c r5:dee5c0ec r4:dee5c0d8
[42487.910119] (0)[848:ndroid.systemui]r3:00000002
[42487.910148] (0)[848:ndroid.systemui][<c0410b90>] (musb_root_disconnect+0x0/0x18c) from [<c04031f8>] (musb_interrupt+0x5f8/0x13d4)
[42487.910163] (0)[848:ndroid.systemui] r7:df1fbf80 r6:00000019 r5:dee5c0d8 r4:c0e8db50
[42487.910208] (0)[848:ndroid.systemui][<c0402c00>] (musb_interrupt+0x0/0x13d4) from [<c05966e8>] (mt_usb_interrupt+0x308/0xbcc)
[42487.910237] (0)[848:ndroid.systemui][<c05963e0>] (mt_usb_interrupt+0x0/0xbcc) from [<c00e96fc>] (handle_irq_event_percpu+0x6c/0x2b8)
[42487.910264] (0)[848:ndroid.systemui][<c00e9690>] (handle_irq_event_percpu+0x0/0x2b8) from [<c00e9994>] (handle_irq_event+0x4c/0x6c)
[42487.910293] (0)[848:ndroid.systemui][<c00e9948>] (handle_irq_event+0x0/0x6c) from [<c00ec410>] (handle_level_irq+0xbc/0x10c)
[42487.910308] (0)[848:ndroid.systemui] r6:c0b0f350 r5:c0b00a50 r4:c0b00a00 r3:00022008
[42487.910350] (0)[848:ndroid.systemui][<c00ec354>] (handle_level_irq+0x0/0x10c) from [<c00e8e20>] (generic_handle_irq+0x50/0xc0)
[... cut ...]
[42487.910548] (0)[848:ndroid.systemui]bfe0: 00000006 bee2c3c0 419462a8 4194d37c 20000010 ffffffff
[42487.910561] (0)[848:ndroid.systemui]machine_shutdown: done
[42487.910571] (0)[848:ndroid.systemui]Reboot:machine restart...
[42487.912875] (0)[848:ndroid.systemui]arch_reset: cmd = NULL
[42487.913169] (0)[848:ndroid.systemui]wdt_arch_reset called@Kernel mode =
[42487.913753] (0)[848:ndroid.systemui]wdt_arch_reset called MTK_WDT_MODE =5c
[42487.914305] (0)[848:ndroid.systemui]wdt_arch_reset called end  MTK_WDT_MODE =22000014

dextercai, 2016-12-25:
here is the last kmsg log. l will catch again on my PC. it seems the sampling rate problem. 

also, why the 1.13(1303) fc on my phone.
maybe, l can catch the log about the fc again.

pavlus, 2016-12-26:
Still need logcat (`*:D`), to see what happens in system and app. It looks like host controller disconnects device, but there is some problem with it, don't want to dive into MTK usb host driver that deeply yet.

I assume, you use HackRF, right?
Do you use powered hub or Y-OTG cable with sufficient power capability?

pavlus, 2016-12-26:
And, since it's issue with usb part, it should be reopened at HackRF lib https://github.com/demantz/hackrf_android
There is a test app, try playing with it and see what happens.

dextercai, 2017-01-08:
l think the hackrf works well. maybe it is because the samples rate is too high. is there any way to limit the samples rate?

dextercai, 2017-01-08:
[image]

dextercai, 2017-01-08:
I'm sorry, but I've got a reply right now. As long as I set samples rate too high it will cant connect to the hackrf. it seems the problem of the  rf app not my os. 
 also, why l just can run the app in the branch scan feature but not in the master. it always fc.

pavlus, 2017-01-09:
It doesn't seem like ok, you got only 3 packets of 16384 bytes, 24k samples at 8Msps is nothing.
Test app hardcoded queue length is 1 second of 15 Msps stream, so it's almost 2 seconds at 8 Msps, your phone have received only 3 packets, that means problem is not with bandwidth, but somewhere between hackrf-lib and HackRF device.

 What's maximum sample rate you can achieve without problems on Test app?

Need more replies from other users with similar phones, it doesn't really looks like

dextercai, 2017-01-10:
l will test on my 2 phones and report later.

dextercai, 2017-01-11:
After my test. Everything is normal on the test app. But the other one is not normal. The application can only receive a few seconds. And then the waterfall stops. Then, RX lights off. If you restart the application and continue to receive. I can't receive anything, RX lights only flash once. If I disconnect the device at this time, The mobile phone will restart.

pavlus, 2017-01-13:
On both phones? Which models?
And once again, can you test your HackRF with same settings on PC?

dextercai, 2017-01-13:
both mtk phones. but on pc it works well. still just can use the ver1.13 but not the other versions.

dextercai, 2017-01-15:
the phones are both base on MTK. and on pc , my hackrf works well.

wissanu020430, 2019-06-06:
https://github.com/demantz/RFAnalyzer/issues/69#issuecomment-499084698

closed 2019-06-06 by dextercai