demantz/RFAnalyzer #47: Tuning Inhancement

BillDunn, 2017-01-17:
Thanks Dennis for a great app. One thing that would greatly enhance this app would be to put something like a fine tune slider at the top of the app for tuning. that way you could click on a frequency and then have a centered tuning bar at the top to slide back and forth to obtain perfect tuning, especially on modes such as SSB.  That tied to an "options addition" for making the steps of 50, 100, 500 would make it perfect.
Thanks,
Wm Dunn

pavlus, 2017-01-18:
You can tune to exact frequency right now, not the same, but may help until implemented.

wissanu020430, 2019-06-06:
https://github.com/demantz/RFAnalyzer/issues/69#issuecomment-499084698

phpbg, 2019-07-30:
Hi, I know devs are standby, but +1 for something like fine tuning.

Tuning on a phone with a small screen is too difficult. Same goes for demodulation bandwidth