demantz/RFAnalyzer #48: App Looking Exact Same As RF Analyze on Google Play

alkimpone, 2017-02-15:
Hello i was on googleplay yesterday and i did see this copy of your program with just diffrent name SDRoid
Have same layout and everything , dont know if you have a copyright on the program itself on googleplay anyways now someone have made a replica of your program for free on googleplay called SDRoid , Just thought i was gonna give a message about it incase you did have some sort of a copyright on the program itself

demantz, 2017-02-15:
Well, RF Analyzer is under GPL license. So basically its fine to fork the code, change it and publish it. I'm not very happy about it though.. It's just a bold copy (I can not see any new/improved features; it's even outdated and lacks some features).

But thanks for mentioning it! Maybe I will leave a commend in the review of SDRoid so that others are informed that it is a clone.

alkimpone, 2017-02-16:
Yeah i find it realy bad as he have just sort of copy everything and you are the one with the real source code and also have insane good instrucs on how to get stuff up and working and all , you ask me i think your app shud been the main app for this project not hes but thats my opinion..
And i tryd add a message myself about that app just was a clone of RF analyze on googleplay , anyways thanks for a amazing project and good work done 👍

closed 2017-02-16 by alkimpone