demantz/RFAnalyzer #50: is the received data on the basebaud?

esgx001, 2017-03-12:
because i wanna do tx on android,so iq file is like this, sinx, cosx,at the basebaud

pavlus, 2017-03-12:
If you mean stored *.iq file -- it has same format as stream from device, samplerate is written in file name and each sample is complex, but representation depends on device. As soon as you can't change rate while recording, assume that your rate is fixed.

For HackRf format is 8-bit signed pairs of [Re Im] ([I Q]), you can find this in converters code.

esgx001, 2017-03-13:
ok,i mean the i and q are at the baseband, i am trying the transmitter, record the voice, interpolate,and fm modulate ,get sin and cos, if i can put the sin,cos,sin,cos like this to the iqfile, thenset any frequency to transmit it.

pavlus, 2017-03-13:
If you play them as baseband -- you'll get signal downmixed from frequency in their name(Hz) with bandwidth of samplerate(Sps).
so, signal recorded from 1Ghz at 1Msps will have frequencies from range 999,5MHz -- 1000,5MHz, but downmixed to -0,5MHz -- 0,5MHz.

esgx001, 2017-03-14:
so you mean i need to do upmixed?but i've read the guide of the hardware, found that it has finished it, the usb got is exactly the baseband.we can use the method of "hackrf.setFrequency(frequency)"to change the frequency.

pavlus, 2017-03-14:
I don't know how you are going to use it, on which hardware, software or anything, just "sharing wisdom" about how it's done. 

It's up to you how to use it, I don't even have HackRF and don't know much about it to recommend something.

And be aware of which signal is recorded, it's everything received from device, everything visible on spectroscope and not the part which selected by squelch window when demodulating.

esgx001, 2017-03-14:
i mean what i need to do when i choose to transmit the FM signal. i've done recording, interpolation, fm modulation, so then? whether i transmit the fm signal is ok? or what i need to do ?

pavlus, 2017-03-14:
Describe what you are trying to achieve and by what means.

wissanu020430, 2019-06-06:
https://github.com/demantz/RFAnalyzer/issues/69#issuecomment-499084698

closed 2025-08-05 by demantz