demantz/RFAnalyzer #51: Hackrf sweep implementation.

W-M-D, 2017-03-19:
Would it be possible to implement hackrf sweep in this app ?

demantz, 2017-03-20:
I will try... its on my TODO list ;)

careyer, 2017-09-19:
+1! Yes, HackRF Sweep Support would be a great benefit. It works awesome on WindowsPC ... however my Workstation is not really portable :-/

wissanu020430, 2019-06-06:
https://github.com/demantz/RFAnalyzer/issues/69#issuecomment-499084698

umer0zerve, 2022-08-12:
2. Is is there is any correct way of reading hack-rf sweep data in python?
I've used complex 64, int8, int16 etc. 

def Read_method3(path):
    iq = np.fromfile(path, np.int8)
    iq = (iq[::2] + 1j*iq[1::2])
    return iq

Int8 works well for hack-rf transfer file (file saved at single frequency). however, its not working well for hack-rf sweep file (file sweep between 500 - 6000 MHz).

umer0zerve, 2022-08-12:
1. is it possible to  extract cell id from hackrf sweep?
Hi All
My task is to extract cell id from hack-rf files. I have two files 
1. hack-rf transfer (file sampled at 1 frequency)
format of the collection:
hackrf_transfer -f {LTE_frequency} -s 19200000 -b 20000000 -n 1728000 -l 16 -a 1 -g 30 -r {path to cfile or bin file}

2. hack-rf sweep (file sampled between 500 - 6000 MHz). 
format of the collection:
hackrf_sweep -a 1 -f 500:6000 -p 1 -l 16 -g 30 N 1 -B -r {output_dir}/sdr/{time}.cfile

I'm able to extract cell id from the hack-rf transfer file which was sampled at a single frequency. However, I'm not able to extract the cell id from hack-rf sweep file. 

is it really not possible to extract cell id from hack-rf sweep?