demantz/RFAnalyzer #52: Works on LAN but not well over public internet

joejoe, 2017-03-26:
I like your app, and thank you for making it $1. I really appreciate it.

Using it on my LAN, no problem.

But using it over an Ample high speed internet connection, it just doesn't work. I get very intermittent audio and sometimes it sounds like the person is speaking slowly.  I do have it behind NAT and port forwarding but I don't think that's the problem.

Is there a way to possibly add a Buffering option in the application so it could catch up with itself possibly?

Is the problem general internet  latency? I'm going from NY-Time Warner to NY-Verizon. Thanks.

pavlus, 2017-03-28:
If you can build app yourself -- try to increment this value https://github.com/demantz/RFAnalyzer/blob/master/app/src/main/java/com/mantz_it/rfanalyzer/RtlsdrSource.java#L83 like to 100, or more. Report on success, please.

wissanu020430, 2019-06-06:
https://github.com/demantz/RFAnalyzer/issues/69#issuecomment-499084698

closed 2025-08-05 by demantz