demantz/RFAnalyzer #55: License

pavlus, 2017-04-03:
While developing fork, I'm using libraries released under [Apache License, version 2.0](http://www.apache.org/licenses/LICENSE-2.0) (right now it's GreenRobot's EventBus and will be using Dagger soon). The problem is, that Apache License v2 libraries are incompatible with GPLv2 licensed projects.

Could you consider a relicensing project under more permissive license (like Apache License v2, MIT, or BSD 3/2-clause)? 

If you want to stick with GPL, it could be GPLv3.

Apache License v2 allows commercial use and explicitly gives permission to the user about patents used. In case some hardware developer or company would like to contribute to the app, but without disclosing its algorithms, it's more likely that they will contribute to Apache Licensed project. Also, it would allow usage of closed-source plugins.

This is not a legal advice, I'm not a lawyer and further research needed. 

I have not licensed my contributions yet, but I'd like to use Apache License.

demantz, 2017-04-04:
One reason I chose GPLv2 was that the FFT.java implementation (https://github.com/demantz/RFAnalyzer/blob/master/app/src/main/java/com/mantz_it/rfanalyzer/FFT.java) is also licensed under GPLv2 (and not explicitly under GPLv2+). Because v2 and v3 are incompatible I think I cannot put my project under GPLv3 and still use the FFT implementation, right?

However, I put RF Analyzer under GPLv2+. So upgrading to GPLv3 shouldn't be a problem as long as the FFT implementation is replaced with a compatible license..

wissanu020430, 2019-06-06:
https://github.com/demantz/RFAnalyzer/issues/69#issuecomment-499084698