demantz/RFAnalyzer #57: Looks there is unwanted squelch

9a4gl, 2017-06-13:
Hi,

when listening HF with very bad antenna signal get so low and I do not hear anything while tuning. Looks like there is squelch implemented. How about deactivating it and listen to noise always ?

pavlus, 2017-06-17:
Which demod?

9a4gl, 2017-06-18:
LSB

pavlus, 2017-06-20:
Most probably, the problem is with squelch tied to the bottom of the FFT display, so you have to zoom out and move it down as low as you need and don't zoom in again, or it will be moved upwards when you will go to the top.

Or you can try to build the app yourself commenting these lines: https://github.com/demantz/RFAnalyzer/blob/master/app/src/main/java/com/mantz_it/rfanalyzer/AnalyzerSurface.java#L669-L673 it will stop trying to keep a squelch within a screen.

9a4gl, 2017-06-27:
Just tried, first suggestion didn't helped. But second one solved the problem. Thanks. Maybe we need a way to disable squelch completely in settings. It is not so common to use squelch for LSB/USB on HF.

9a4gl, 2017-07-04:
Looks there are more problems with squelch with narrow channels, take a look at this fix
https://github.com/9a4gl/RFAnalyzer/commit/4a76e37dc4b40ae411f9356a0d52e323fb84aa28
chanStart and chanEnd are same if channel is not wide enough

wissanu020430, 2019-06-06:
https://github.com/demantz/RFAnalyzer/issues/69#issuecomment-499084698