demantz/RFAnalyzer #61: apk for andriod

cinaeds, 2018-06-15:
I don't have WIFI or cell phone access in the short term so I can't install RFAnalyzer using either Amazon or Google Play. 

I did install the version from F-droid on my Android since I can download the apk from there with any web browser - namely the one on my desktop - but it's an old version. 

If you email the zip file of the latest version of the apk, I'll donate 30 EUR  (or 34.7 USD) using Paypal. 

If you agree - or you can think of a better way then using email - let me know, and you can send it after I donate 30 EUR at http://tech.mantz-it.com. 

-- Cinaed

pavlus, 2018-06-18:
The latest version (1.13) is available for download from this repo, just in the root directory. 
https://github.com/demantz/RFAnalyzer/blob/master/RFAnalyzer.apk

wissanu020430, 2019-06-06:
https://github.com/demantz/RFAnalyzer/issues/69#issuecomment-499084698