demantz/RFAnalyzer #68: Spectrum scaling compressed in comparison to sdrangel and gqrx

jbvoelker, 2019-04-20:
I am currently using RFAnalyzer on a Tablet connected to a RTL-SDR as part of my Es'Hail-2 receiving setup.
This works great, but the signal levels on the spectrum view seem to be compressed compared to the spectrum display in gqrx and sdrangel.
For example, the beacon signal which peaks ~ 25 dB above the transponder noise with sdrangel and gqrx peaks ~ 15 dB above the transponder noise in RFAnalyzer.

I have built an apk with line 217 in AnalyzerProcessingLoop.java changed to
`mag[targetIndex] = (float) (10* Math.log10(realPower + imagPower));`
and the resulting spectrum display now seems to be consistent with gqrx and sdrangel.

Can somebody confirm the issue, or is it just a problem with my RFAnalyzer settings?

wissanu020430, 2019-06-06:
https://github.com/demantz/RFAnalyzer/issues/69#issuecomment-499084698

demantz, 2025-08-03:
You are absolutely correct. Thank you very much for reporting this. It is fixed in the new RF Analyzer 2.0 version which releases soon..

closed 2025-08-03 by demantz