demantz/RFAnalyzer #78: Frequency is out of valid range

llcoder00, 2021-03-08:
Using Samsung S10 with OTG adapter with rtl-sdr v3 dongle. Selected AM and frequency of 1MHz, but won't let me select any AM frequencies.  Error message is "frequency is out of valid range.

haveyouwantto, 2022-12-20:
I figured out, just edit line 90 in `com.mantz_it.rfanalyzer.RtlsdrSource`
replace
```java
24000000l,	// R820T
```
to
```java
0,	// R820T
```
and it should work.

khaytsus, 2025-09-25:
> There will be a new RF Analyzer 2.0 version soon which is almost a complete rewrite of the app. This will have the option to disable all frequency restrictions so that arbitrary tuning for RTL-SDR is possible. Nevertheless, thank you for the pull request!

Did this ability to disable tuning limitations happen in 2.x and I'm missing it?  I have an RTLSDR Blog v4 and it's still showing a lower limit of 24MHz.

haveyouwantto, 2025-09-30:
Enable allow out-of-band frequency in settings