demantz/RFAnalyzer #79: Headphones as antenna

paulnaude, 2021-05-05:
Hi
This app is exactly what I was looking for but don't need the accuracy I suppose the required extra hardware brings that is required. Can't it be modified to use the received FM frequencies from an ordinary headset (which is required as an antenna) to scan the range of frequencies and present the same rainfall plots?
It would help me find the empty FM bands along my regular route to use my Bluetooth FM transmitter in my car without interference.

penguin359, 2022-01-21:
You need more than an antenna, you need an FM receiver and I am not aware of any Android phones that ever came with an FM receiver in hardware. The old Nokia phones that I am aware of that could receive FM radio when using the headphones had the FM receiver inside the big plastic button they came with. They would feed the audio directly into the headphones and the phone itself was just a power source and interface for controlling it.

Also, if there was a real FM receiver on a phone, it would likely only give you the demodulated audio after narrowly tuning to one station instead of the wider RF spectrum. To do what you need, you should really just go for an RTL-SDR dongle which can be quite cheap such as here:

https://www.amazon.com/JahyShow%C2%AE-DVB-T-RTL2832U-Receiver-Compatible/dp/B01H830YQ6/ref=sr_1_8?crid=2FT7KPY8RK6KJ&keywords=rtl-sdr&qid=1642801509&sprefix=rtl-sdr%2Caps%2C213&sr=8-8

However, if you are going to spend a little money, I'd go with something slightly more rugged like this one:

https://www.amazon.com/RTL-SDR-Blog-RTL2832U-Software-Defined/dp/B011HVUEME/ref=sr_1_3?crid=2FT7KPY8RK6KJ&keywords=rtl-sdr&qid=1642801636&sprefix=rtl-sdr%2Caps%2C213&sr=8-3

Besides that, you will probably also need an adapter to USB-C or USB Micro depending on what your phone has which are also pretty cheap:

https://www.amazon.com/Syntech-Adapter-Thunderbolt-Compatible-MacBook/dp/B07CVX3516/ref=sr_1_3?keywords=usb-c+to+usb-a+adapter&qid=1642801726&sprefix=usb-c+to+usb-a+%2Caps%2C149&sr=8-3
https://www.amazon.com/Ksmile%C2%AE-Female-Adapter-SamSung-tablets/dp/B01C6032G0/ref=sr_1_3?crid=3K0BXZYBAZ4J3&keywords=usb+micro+otg+adapter&qid=1642801790&sprefix=usb+micro+otg+adapt+e%2Caps%2C138&sr=8-3

paulnaude, 2022-01-22:
Thank you for your suggestions. In the meantime I did get a dongle and
antenna  and used it with an USB to USB-C adapter on my phone but the
adapter gets hot quite quickly and then the app looses the connection. I
suppose using a laptop instead is the solution.
Or does it point to a faulty adapter?

Regards

penguin359, 2022-01-24:
From my experience, some of those RTL-SDR dongles do get quite warm typically, but I don't normally have them loose connection. It could be your phone is unable to handle it's power requirements or it could be a more inefficient design of dongle as there are quite a few vendors these days. Does it have a metal enclosure?

Another option might be to get a powered USB-C hub. That will also allow your phone to charge while also powering the dongle. This is one I've used in the past, but there are quite a few options as well:

https://www.amazon.com/LENTION-Compatible-2021-2016-ChromeBook-Multiport/dp/B098JTWV78/ref=sr_1_19?crid=3B08QEWWGST06&keywords=usb-c+hub+power&qid=1643006962&sprefix=usb-c+hub+power%2Caps%2C148&sr=8-19

paulnaude, 2022-01-29:
Thanks for the tip!

closed 2025-08-05 by demantz